samphonic/Moppy2 #185: Motor  spins all the way to the end??

Tofu-86, 2023-03-06:
This may be because i'm using a 6V battery pack to power the floppy drive. I just connected the positive and negative to the corresponding pins and followed you tutorial for the arduino setup. The songs originally worked but now the motor spins and the reader thing gets stuck at the end.
If you need a video or picture ill be happy to send one.

Sammy1Am, 2023-03-06:
If the motor is spinning, it sounds like you might have one of the pins on the floppy drive side of things mixed up.  Maybe double-check which pins you're connected to (and that you're counting from the correct side).  A photo of the connector on the back of the floppy drives with wires connected would allow me to count too if you're unsure.

I *suspect* 6v is probably fine (or at least shouldn't be causing the problems you're seeing).

Tofu-86, 2023-03-15:
Ah i forgot, I did actually fix the issue right after, turns out the batteries were unstable and weren't supplying a constant 6v when I tested them, i switched them out and it worked like a charm!

closed 2023-03-15 by Tofu-86