samphonic/Moppy2 #187: Alternative power supply and drive not stepping.

Sidges, 2023-05-14:
I'm not very proficient in electronics and stuff like this so trying to create this has been quite a fun challenge. I've come into a question and an issue:

1) Sadly I do not have an ATX power supply and was wondering if it's possible to simply use a 9V battery and some resistors instead?
2) Whenever the start up noise or music is supposed to be playing the head isn't moving and instead I can hear a very faint clicking noise from the drive.

Might these issues be connected?

Sammy1Am, 2023-05-14:
1. It *might* be possible to use a 9v battery to power the floppy drive, but you're probably better off trying to find a 5v wall adapter from something and using that instead.  Using resistors would be tricky because the current drawn by the drive isn't constant; you could definitely use a DC-DC converter to convert from 9v to 5v, but unless you already have one, you're more likely to have a 5v wall adapter.
2. The semi-rhythmic ticking is usually a sign that the step and direction wires are swapped.  Try switching them and see if that helps.  You can also try confirming you have the right ones by manually connecting the wires to ground to step the drive by hand (just to make sure the wires are right on the floppy drive).

Sammy1Am, 2023-05-16:
🎉 glad you got it working!

closed 2023-05-16 by Sammy1Am