samphonic/Moppy2 #190: Matainted

OverloadedTech, 2023-06-10:
Hello,

Great project
I think i will try it out after i make sure to have all the necessary hardware
In any case: is the software still updated and maintained with updates?

Thanks
Luca

DJthefirst, 2023-06-10:
It is not actively being developed but the project is stable and Sammy is around to provide support. I am working on a similar program but completely different structure to improve upon some of moppys shortcomings. Though that will not be ready for a while yet. If you do encounter problems just report them here and someone will be willing to help out.

Sammy1Am, 2023-06-10:
Hi there,

Thanks, you should definitely give it a try, it's pretty fun.

The short answer is: no.  The longer answer is: there haven't really been any updates, but it's technically "maintained" in that I'm not aware of anything that's currently broken, so you should be good to go.

I'm always happy to consider PRs if anyone does add any new features or updates, but it hasn't really come up much.

OverloadedTech, 2023-06-10:
Ok, thanks for the reply
There it the possibility to do also this thing with some HDD's (like what Paweł Zadrożniak did that use Floppy Readers and HDD)?
Let me know

DJthefirst, 2023-06-10:
So when driving a floppy drive you use two wires. One to pulse the drive head at audio frequency and the other to set the direction. On hard drives you probably can’t drive it just from the external connectors like you can floppy dives as computers access hard drives at a higher level. I have opened up a hard drive and on the motor soldered two wires. One goes to Gnd and the other goes to one of the pins that pulses at audio frequency. I think you can also program an Arduino to be of a type that only has PWM audio pins.

DJthefirst, 2023-06-10:
Pretty much any motor can be driven to play music you might just need transistors or a motor driver to boost the power.

OverloadedTech, 2023-06-11:
Ok
Thanks for the help

OverloadedTech, 2023-06-11:
I already was able to find an Arduino and the jumper wires but i am afraid I couldn't find the appropriate power supply and floppy Reader
Could someone share the link to the working components?

OverloadedTech, 2023-06-12:
Let me know

OverloadedTech, 2023-06-12:
Hello,

Yes, for the floppy drives i am ok but i am afraid i would have issues regarding the ATX Power Supply
Is an ATX Power Supply with similar cables, but i am not sure if that is compatible (see [image](https://i.ibb.co/1mBr1dX/Cattura.png))
Let me know if this is compatible or if there are any kinds of issues

Thanks

DJthefirst, 2023-06-13:
You can't directly connect it but power is power. All you need to do is connect a black and red wire to those of the floppy drive. Wether it be solder or jumpers. A female to male Arduino wire should work just might get a bit warm. You shouldn't need 12v just lookup floppy drive power connector to se the pinout everything is color coded.

OverloadedTech, 2023-06-13:
Can't i use directly the Arduino to provide power? I guess no, right?

DJthefirst, 2023-06-13:
The Arduino at best can put out 200ma and the Floppy takes 2A at best you would brown out the supply and it would crash at worst you'd kill the Arduino.

OverloadedTech, 2023-06-14:
> Hello,
>
> Yes, for the floppy drives i am ok but i am afraid i would have issues regarding the ATX Power Supply Is an ATX Power Supply with similar cables, but i am not sure if that is compatible (see [...]

I was wondering if the power supply image seen in the comment quoted up here could be connected like in the Sammy1Am's Video ([video](https://www.youtube.com/watch?v=D2iswCnV9ec))

DJthefirst, 2023-06-15:
Yes just you don't have the connector but if you can connect 5v red and gnd where the connector is in the video it will work.

OverloadedTech, 2023-06-15:
Thanks @DJthefirst for the clarification
Just to end this discussion can [this](https://i.ibb.co/C1WbxKW/IMG-20230615-140531-472.jpg)  little contraption i found provide power to the floppy directly connecting the USB to the wall or to a PC?
It should power one single unit and i think it is ok

DJthefirst, 2023-06-15:
It should work. The 2A rating is a maximum so most of the time you are not drawing that. I just wouldn't recommend plugging it into a computer.

OverloadedTech, 2023-06-15:
Nope, i think i will use directly a wall adapted (i live in Europe)

OverloadedTech, 2023-06-16:
> Nope, i think i will use directly a wall adapted (i live in Europe)

Right?

DJthefirst, 2023-06-16:
Yes.

OverloadedTech, 2023-06-16:
Ok fine
Thanks a lot to everyone who joined this conversation. Now i am going to close it