samphonic/Moppy2 #191: Description of the device

Besteciler, 2023-06-12:
Hello
I looked at the code, everything is very cool.
Almost every command has an explanation in the code.
But I'm interested in the description of the operation of the device after it is turned on.

Could not help with a description of the work after switching on and after downloading the melody. What is actually happening at this moment.
And is there a block diagram of the algorithm?

DJthefirst, 2023-06-12:
I’ll just write out the general concept and can go into a deeper explanation later if you want. Basically the Song is in MIDI format so java has a lib that plays the song in time. when it is playing notes are generated and sent as note on and off commands. MIDI has 16 channels so these notes are messages with an assigned channel and cod. I wold recommend looking at the midi protocal. After that the messages go through moppys mappers bc floppy drives can only play one note at a time so chords are a problem. Notes are sorted by channel distributed with round robin and removed if they are out of range etc. This is mapping instruments to physical floppy drives. Then it is sent as new moppy msg to the drive which was encoded. It gets a device ID for each microcontroller and instrument Id. Then the microcontroller has an interrupt and counter that will toggle the pin at the frequency of the note from that lookup table.

DJthefirst, 2023-06-12:
If your interested in how moppy works I would also suggest you look at my project with uses the same general [Idea](https://github.com/DJthefirst/Mechanical-Midi-Music/tree/main/ESP32_Client_V1.0/src). The difference is instead of being split between a java server and microcontroller I use ESP32 to do all the processing on the micro controller. The communication fully uses the midi protocol and instead of having a mapper I have virtual distributors that take a MIDI channel and distribute it to a pool of instruments(floppy Drives)

closed 2023-06-14 by Besteciler