samphonic/Moppy2 #194: Pinout for 74HC595 using ESP8266

DJPatron, 2023-06-18:
Please help. How to connect 74HC595 shift register to ESP8266?
I never able to find correct pinout.

DJPatron, 2023-06-18:
https://github.com/Sammy1Am/Moppy2/assets/67385488/019b24f6-9289-443d-831d-013145b48570

Sammy1Am, 2023-06-18:
If you're using a shift register for floppy drives (i.e. you've uncommented [INSTRUMENT_SHIFTED_FLOPPIES](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/MoppyConfig.h#L19) in the config file), then aside from the latch pin (defined as pin `2` [on this line here](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/MoppyInstruments/ShiftedFloppyDrives.h#L19)), it's using the default SPI pins for the ESP8266.  It varied a bit based on the board you have, but you can find [some info here](https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/) (or by Googling your specific board).

closed 2023-06-25 by DJPatron