samphonic/Moppy2 #195: ...Android 🙈

cookins, 2023-06-22:
Hey there.
Last time i visited this project, it was version 1 lol.. just tested moppy 2. It's great that project still live! But seems still lacking to add multiple midis at once... Anyway -
I'm sure I'm not first, who considering to have version of gui for Android.. or can be quite portable, without pc or raspberry, but via Android. Is it considered in future plans?
Ty.

DJthefirst, 2023-06-23:
Define lacking multiple midis. You can use a Daw or something like loopmidi to have multiple midi input and outputs. 

As for future plans I don't know of any plans Sammy has for Moppy 3 but he has still maintained support for Moppy 2. So any new features would mostly be community led. Sammy might know if someone has made an android port for Moppy. I remember something about one but I think that was for Moppy 1.

I am working on a project inspired by Moppy but completely changing the core structure to allow for more flexibility. [Mechanical Midi Music](https://github.com/DJthefirst/Mechanical-Midi-Music). It uses the raw midi protocol instead of proprietary messages and can be controlled just by going to a web browser (Win/Linux/Mac). It also has the goal of supporting headless operation where you can control instruments directly from a midi controller or keyboard. As for android support I'd be willing to look into it at a later date.

It's not fully operational yet but I hope to have an initial version ready by the end of August. The only downside is it runs on Esp32s. I have to work on porting it to lower power arduinos or make a lite version.

cookins, 2023-06-23:
I mean, that stock GUI don't have functionality, to add to playlist multiple midis at once (folder with midis). Just one by one.

Sammy1Am, 2023-06-26:
Actually, there **is** [an Android app ](https://github.com/NoahR-ATC/MoppyAndroid) available!  @NoahR-ATC made, from the looks of it, a fairly full-featured one that includes the playlist functionality you were looking for and is already available in the Google Play app store (see link in his repository).

Definitely worth checking out I'd say.

cookins, 2023-06-26:
Oh, great! Thank you