samphonic/Moppy2 #196: Windows Java Runtime Error

Wev-Oppy, 2023-07-19:
When running the MoppyControlGUI.bat in command prompt all i get is this error

Exception in thread "main" java.lang.UnsupportedClassVersionError: com/moppy/control/MoppyControlGUI has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClass(Unknown Source)
        at java.security.SecureClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.access$100(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.launcher.LauncherHelper.checkAndLoadMain(Unknown Source)

 Wondering if anyone has a solution to this.

Sammy1Am, 2023-07-19:
You can search on Google or through previous issues for more details, but
in short you need to have a newer version of Java installed.

Wev-Oppy, 2023-07-19:
Turns out the version was too low.
Thanks!

closed 2023-07-19 by Wev-Oppy