samphonic/Moppy2 #198: Floppy drive doesn't play music or startup tone

gusmuncher, 2023-09-10:
When i press the reset button on my arduino the drive head moves a mm backwards and then does nothing.
i tried playing a song and nothing happens.
i tap the wires together to see if the step pin and direction pin work properly and they work fine.
i try to rearrange the wires on the arduino and i get the same result.
i try to upload moppy classic to the arduino and i get the same result
i get no errors in the arduino IDE console btw 

i have no idea what to do now and i would really appreciate it if someone could help me with this.
i am using: a 3rd party arduino uno, 300w ATX power supply, a samsung floppy drive and some M/F jumper wires 
here are some pictures

[image]
[image]
[image]
[image]

Sammy1Am, 2023-09-11:
Take a close look at the output of the Arduino IDE (or screenshot it here if it seems weird), but there's a decent chance something failed while trying to upload the code.  The Arduino IDE is very subtle about telling you it failed to upload the code unfortunately.

gusmuncher, 2023-09-11:
[image]
this is all that appears

Sammy1Am, 2023-09-12:
Ah, after closer inspection of the images (thank you for sending those), you have the step and direction pins swapped.  The red wire should be on Arduino pin 2, and the black on 3.

Sammy1Am, 2023-09-21:
The videos are sort of helpful as the two clicks seem to suggest that:

- The Arduino probably is properly flashed (otherwise shouldn't be sending two signals on reset).
- The step and direction pins are backward.

Now I know you already tried switching them around once, but the startup tone probably causes the head to reverse direction a couple times, and each of those would be interpreted as a single step if the pins were swapped.  So maybe try switching them back?  (Sorry, not super helpful, I know)

You said manually touching the wires together works though, which would seem to suggest that it's NOT a floppy drive issue and is an Arduino issue.  But if things uploaded successfully to the Arduino, then the Arduino is also fine and you don't have an issue... but something is clearly not working 😜 

Another thing to check: when you open the MoppyControlGUI and select the correct COM port, does the Arduino information show up in the upper right of the GUI?

This _shouldn't_ be the case, but another thing to try would be trying the other Drive Sel pin (like if you're using B use A or vise versa).  I have seen at least one drive that was hardwired to be the one I wasn't expecting.  Again, if manually tapping wires works as expected then this isn't an issue, but if you're looking for other things to try.

gusmuncher, 2023-09-22:
Selecting the right COM port shows this (sorry about the quality)

[image]

playing a midi file then stopping it makes the floppy drive make a sound but that's about all that happens.
**here is a video of the drive doing its thing when i either reset it or stop it while the GUI is playing a midi file**

https://github.com/Sammy1Am/Moppy2/assets/144593320/1ad38661-5557-4f18-a6b0-8f330427b371

I tried to use another drive select pin (pin 14 i believe) and it did the same thing.
Also I dont know if this helps at all but  while playing a midi file the arduino's LED stays red and doesn't flash.

Sammy1Am, 2023-11-04:
Phew, sorry, life got pretty busy for a bit.  Which MIDI file are you trying?

Because it's showing up in the UI (thank you for the screenshot) you've got everything uploaded on the Arduino side, so that's fine.  And the reset noise when you reset or stop seems to suggest that the wiring is also correct (you can triple-check by manually moving the read head out a ways and see if the reset button brings it back to the beginning).

So all I've got left for guesses is that the MIDI file is weird, or the notes aren't being mapped correctly or something.

gusmuncher, 2023-11-05:
No worries man. I tried using two of the three midi files that were included in the moppy software folder which are "turret opera" and "tetris" 

but i have one thing to point out. When i was trying to open the java file for the first time it kept crashing so i installed a software (forgot what it's called) which had an older version of java and only then was i able to launch it.
so maybe it's that.

Sammy1Am, 2023-11-05:
Both of those MIDI files should have worked.  (Just making sure)

Unfortunately I'm a little bit stumped.  Seeing the Arduino in the GUI means the firmware flashed to the Arduino successfully, and the computer is communicating with it.  The reset functionality working when you press stop or reset suggests that your floppy drive is probably wired correctly.  The MIDI files were my last guess, but both of those should have notes on the first channel that would go to that drive.

I guess one last thing I can think of: could you post a screenshot of the whole MoppyControllerGUI window?  Maybe something got messed up with the mapper scripts that were set up. 🤷‍♂️

Sammy1Am, 2023-11-06:
Okay, so, this seems weird, but could you try:

1. Switching the Condition menu to "Custom" and changing the text field to `true`.
2. Switching the Sub Address menu to "Custom" and changing the text field to `1`.

The default Condition is filtering on channels 1-4, but then the Sub Address does a +1, which seems to suggest that c should be 0-3.  So... it's possible I screwed that up forever ago and no one's caught it somehow?  Seems weird, like I said, but maybe that's what's going on.

Looks like Adoptium is some kind of Java runtime.  If that were the problem I would expect more to be broken, so I think it's probably fine.

gusmuncher, 2023-11-07:
It works now! thank you so much for your help! 
I managed to connect all 3 drives and all of them work :D

Sammy1Am, 2023-11-07:
Hooray!

That's crazy that apparently that mapping script has been wrong this whole time but no one else has noticed.  I'll file a bug for it and hopefully fix that eventually.

closed 2023-11-07 by Sammy1Am