samphonic/Moppy2 #199: Compilation error: 'TIMER_RESOLUTION' was not declared in this scope

Del161, 2023-09-14:
Keep getting this error when trying to upload the moppy-arduino to my esp32.
i followed the tutorial and downloaded the needed libraries.
does anyone have an idea to what is going on?

Sammy1Am, 2023-09-15:
Hmm, `TIMER_RESOLUTION` is defined in `MoppyInstrument.h`.  How are you downloading the Moppy project?  Are you unzipping it after downloading?  What IDE are you using, and how are you opening it?

Nughu, 2023-11-22:
same problem here. downloaded from releases page, unzipped in folder, opened Moppy2-Arduino.ino in Arduino IDE.

Sammy1Am, 2023-11-22:
Hmm, I'm unable to test it myself at the moment.  Could you send me a screenshot of your Arduino IDE when the error occurs?

Nughu, 2023-11-25:
Actually, I got it working by using the platformio ide in visual studio instead.

Sammy1Am, 2023-11-27:
Well, @Del161 , if you get a chance to send me a screenshot maybe I can sus out what's going on.  It's possible the Arduino IDE changed somehow and doesn't support the layout of this project anymore.

Del161, 2023-11-27:
@Sammy1Am Ill try to send a screenshot as soon as i have time, what specifically would you like to see?

Sammy1Am, 2023-11-27:
Mostly just the state of the Arduino IDE; which files do you have open /
tabs.  Is there any other output before the error when you try to compile,
etc.

I don't know specifically if there's something I'm looking for, just trying
to figure out what might be going on.