samphonic/Moppy2 #201: code not compilation

shadow54658, 2023-10-15:
Using library TimerOne at version 1.1.1 in folder: C:\Users\Oleg\Documents\Arduino\libraries\TimerOne 
Using library SPI at version 1.0 in folder: C:\Users\Oleg\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\SPI 
exit status 1

Compilation error: exit status 1

Sammy1Am, 2023-10-15:
Hmm, is there any output from the complication that includes an error?  Which IDE are you using?

shadow54658, 2023-10-16:
Thank you so much for responding, I use Arduino IDE 2.2.1, so to speak, I am not very experienced in this topic, I may not have noticed a way out of the situation in the logs, I immediately apologize for the translation because I use a translator from Russian to English, the following text is in the logs
FQBN: arduino:avr:uno
Using board 'uno' from platform in folder: C:\Users\Oleg\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Using core 'arduino' from platform in folder: C:\Users\Oleg\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6

Detecting libraries used...
C:\Users\Oleg\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Users\Oleg\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\Oleg\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\standard C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\Moppy2-Arduino.ino.cpp -o nul
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\main.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\MoppyInstruments\EasyDrivers.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\MoppyInstruments\FloppyDrives.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\MoppyInstruments\L298N.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\MoppyInstruments\MoppyTimer.cpp
Alternatives for TimerOne.h: [TimerOne@1.1.1]
ResolveLibrary(TimerOne.h)
  -> candidates: [TimerOne@1.1.1]
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\MoppyInstruments\ShiftRegister.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\MoppyInstruments\ShiftedFloppyDrives.cpp
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\MoppyNetworks\MoppyMidi.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\MoppyNetworks\MoppySerial.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\MoppyNetworks\MoppyUDP.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\main\MoppyInstruments\EasyDrivers.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\main\MoppyInstruments\FloppyDrives.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\main\MoppyInstruments\L298N.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\main\MoppyInstruments\MoppyTimer.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\main\MoppyInstruments\ShiftRegister.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\main\MoppyInstruments\ShiftedFloppyDrives.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\main\MoppyNetworks\MoppyMidi.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\main\MoppyNetworks\MoppySerial.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\main\MoppyNetworks\MoppyUDP.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\main.cpp
Using cached library dependencies for file: c:\Users\Oleg\Documents\Arduino\libraries\TimerOne\TimerOne.cpp
Using cached library dependencies for file: C:\Users\Oleg\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\SPI\src\SPI.cpp
Generating function prototypes...
C:\Users\Oleg\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Users\Oleg\AppData\L [... cut ...]
C:\Users\Oleg\AppData\Local\Temp\arduino\sketches\FCE574BD4F7FA3E03B7491CC001A11C7\sketch\src\MoppyNetworks\MoppySerial.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status

Using library TimerOne at version 1.1.1 in folder: C:\Users\Oleg\Documents\Arduino\libraries\TimerOne 
Using library SPI at version 1.0 in folder: C:\Users\Oleg\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\SPI 
exit status 1

Compilation error: exit status 1

shadow54658, 2023-10-16:
I solved the problem, I just needed to download Arduino IDE 1.8. Thank you

closed 2023-10-16 by shadow54658

Sammy1Am, 2023-10-17:
Glad you got it working!