samphonic/Moppy2 #203: MoppyControlGUI.bat crashes

OlegMoore1488, 2023-11-05:
Hi, I have a problem with moppy2.
When starting the bat file, it crashes, I did everything as written in the guide.

Sammy1Am, 2023-11-05:
Try running the .bat file from the command line so you can see which error
is being reported.  Most likely you either need a newer version of Java, or
JAVA_HOME is not set correctly.

OlegMoore1488, 2023-11-05:
[image]
here is the error itself

Sammy1Am, 2023-11-05:
You are attempting to run Moppy with Java 8.  You need at least Java 11 (but even 11 is pretty old, you should just use the latest release).

closed 2023-11-05 by Sammy1Am

OlegMoore1488, 2023-11-05:
installed the latest version of Java (21) it still does not start
[image]

Sammy1Am, 2023-11-05:
Java 8 is still installed and is either set in JAVA_HOME or in your path variable.  Sometimes uninstalling Java 8 will resolve this, though you then might have an error about JAVA_HOME not being set and will need to set it manually anyway.

OlegMoore1488, 2023-11-05:
I deleted all old versions of Java, leaving only Java version 21, but it still does not start
[image]
what else could be the problem?

Sammy1Am, 2023-11-05:
You missed a version of Java 8.

Try `echo %JAVA_HOME%` in a command prompt and let me know what the output is.

OlegMoore1488, 2023-11-05:
[image]
I don't know what the problem is, I deleted the 8th version of Java completely, but in the console it says that it still exists, but if you go down the path, there is nothing
[image]

Sammy1Am, 2023-11-05:
Weird.  There's still a Java 8 somewhere because Java 8 is what's outputting the error message you're getting (it would say something other than `52.0` if it were not Java 8).

Perhaps try setting your `JAVA_HOME` variable to point to the Java 21 installation.

OlegMoore1488, 2023-11-05:
i already fixed everything, everything works perfectly.
thanks for the help