samphonic/Moppy2 #206: ControlGUI sends mdi data to serial, board serial led indicates data comes but the floppy motor doesn't move

stilyan, 2024-03-19:
System is Win10
Using Moppy 2.0
Tried Uno and Nano (chinese clones) to the same result
On startup or board reset the floppy indicates with a tone
Set baud rate 9600 and picked correct COM (confirmed via hardware manager)
In the upper right corner there is no indication that board is picked by the GUI software
When mdi starts the led of serial on board starts blinking
Nothing on the floppy , no movement, nothing

When using old Moppy (classic) INO on the board + the GUI 2.0
Again in upper right corner empty but the floppy motor starts moving , however , no melody , just some weird movements back and forth

stilyan, 2024-03-19:
Also tried testing drives by shorting pins as per guide video , motor moves back-forth indicating no hardware errors in drives

Another think I tried is connecting two drives to pins 1,2,3,4 and only one of the drives work (with old Moppy(classic) INO + GUI 2.0) , the second doesn't move
On board reset however, both drives make the little 'tone'.

Sammy1Am, 2024-03-19:
I don't think you'll be able to successfully use the MoppyClassic embedded
code with the 2.0 GUI.  Please stick to using the newer embedded code as
well and let me know what happens.

Also, just want to confirm that "pins 1,2,3,4" means 2,3,4,5 (since 0 and 1
are COM pins)

stilyan, 2024-03-20:
Using the classic INO on the board with GUI 2.0 was just an experiment, to see if I would get something out of it.

The test setup currently is:
Floppy 18-DIR, 20-STEP are on board pins 3, 2
12-SEL1 is grounded with 9-GND
17-GND goes to board GND pin

Moppy2.0 INO uploaded to the board, no errors. The avrdude while uploading determines the board signature 0x1e950f (probably m328p). As mentioned before, both boards (an Uno and Nano) are clones, non-originals. They use the CH340 driver.

Moppy2.0 GUI opens via .bat
MDI file loaded, correct COM selected (checked via hardware manager), baud rate set to 9600.

When MDI starts, the serial led on board indicates activity but floppy motor doesn't move.
The column at the upper right corner of the GUI doesn't show anything, its grayed.

Please let me know if more details are needed. I can also take pictures of the setup / screenshots of the GUI if necessary.

Sammy1Am, 2024-03-20:
Thank you for the additional information, it sounds like things are pretty much set up correctly.  A couple notes:

- Arduino clones are fine, no worries there.
- Pin connections seem fine.
- With 2.0 INO, on reset, does the first (pins 2,3) drive play a little 4 note melody (as [at 5:00 in this video](https://youtu.be/D2iswCnV9ec?si=GD9Uk5kULAX9ttiq)?)
- If the melody plays, the firmware and wiring are correct.
- If the melody plays but the Arduino doesn't show up in the upper right of the GUI, there's some sort of COMs issue
- The serial LED on the board will flash when data is sent to the board, even if there's no code on the board.  So that's only really an indicator that your PC is trying to communicate, but doesn't mean it was successful.
- If the melody doesn't play, something has gone wrong with the INO upload or configuration (or your wiring is wrong, but that seems unlikely at this point)

stilyan, 2024-03-20:
> With 2.0 INO, on reset, does the first (pins 2,3) drive play a little 4 note melody (as [at 5:00 in this video](https://youtu.be/D2iswCnV9ec?si=GD9Uk5kULAX9ttiq)?)

When reset button is pressed the floppy makes a little sound but doesn't move the motor forth and backwards as shown in the video. This behavior is the same with another floppy as well, doesn't seem to be hardware issue with the floppy itself.
Made a video, https://www.youtube.com/watch?v=0Uey9J4fb30

When I'm playing around with the fires, shorting 18-DIR with 17-GND and tapping the 20-STEP the motor moves forward
Then tapping 20-STEP with 17-GND moves it backwards.
So the wiring and drive seem OK.

> If the melody plays but the Arduino doesn't show up in the upper right of the GUI, there's some sort of COMs issue

Not sure if what I get on my setup considers as the 'melody', but since the board was hooked to the USB of a docking station, I tried the USB on the laptop itself. COM number changed when cable was switched but the same result, the GUI didn't show the board.

stilyan, 2024-03-20:
Managed to get the motor moving with simple piece looking like so
void loop() {
```
digitalWrite(dir, LOW);
for (int i=0; i<= 5; i++) {
digitalWrite(step,HIGH);
delay(1000);
digitalWrite(step,LOW);
}
delay(3000);
digitalWrite(dir, HIGH);
repeat of the for...
}
```
where step = 2 ; dir = 3 (the corresponding pins)

the thing movies back and forth with this
so this tights down to the Moppy2.0 code

stilyan, 2024-03-21:
Got it working.
The code was uploading, the startup melody didn't play because by default in config its defined false.
I left the baud rate 57600 (as by default) instead of setting it 9600 in the GUI, checked the COM checkbox.
Few seconds later on the upper right the device address showed up, I loaded MDI & it started running as if by miracle.

Now is play time, need to hook the 4 more floppies I got and start searching for more to make an orchestra.

Thanks for your help and the work on this repo, really appreciate not having to write this up from scratch myself.

closed 2024-03-21 by stilyan

Sammy1Am, 2024-03-21:
Glad you got it working!

The startup melody *should* default to true though:
https://github.com/Sammy1Am/Moppy2/blob/master/Microcontroller/Moppy2-Arduino/src/MoppyConfig.h#L32

Maybe double check you're on the latest version, or let me know where you found it was disabled (maybe I've missed something)

Sammy1Am, 2024-03-21:
Oh no!  Just downloaded the zip file and it's false there!  😕 What?!

I'll have to fix that.