samphonic/Moppy2 #209: ESP32 support

GoldenMelky, 2024-03-25:
it's the first time that i use platformio, so i don't really know how to use it. I changed the env and board to esp32dev, platform to espressif32 and added two libraries, but when i compile it i get this error
`'TIMER_RESOLUTION' was not declared in this scope`
does someone alredy tried to use Moppy with an ESP32?

(sorry for bad english)
i'm using linux btw.

[image]

Sammy1Am, 2024-03-26:
Hrm, this might be related to a recent change merged to add ESP32 support (but apparently accidentally removing it, haha).  Try replacing [these lines here](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/MoppyInstruments/MoppyInstrument.h#L22-L26) with just the following line:
`#define TIMER_RESOLUTION 40`

That should ensure that it's defined.

DJthefirst, 2024-03-26:
Once you get it working on Esp32 try lowering the Resolution value. I am using different software but can get down to around 10 before it crashes. The lower the value to more accurate the notes and 4x is a major improvement for higher notes.