samphonic/Moppy2 #210: Issue with MoppyControlGUI

yarfw, 2024-04-12:
[image]
[image]
[image]
pls help

Sammy1Am, 2024-04-12:
Huh, well, this will be the first time I've recommended trying this, but: maybe try an older version of Java?  But not much older, just maybe try 21.

The error is deep inside the JRE libraries / graalvm, so it's not something directly in Moppy, but it's possible something has changed / updated that broke since the code was written (or since version 22 is less than a month old, could be a bug?)

yarfw, 2024-04-12:
> Huh, well, this will be the first time I've recommended trying this, but: maybe try an older version of Java? But not much older, just maybe try 21.
>
> The error is deep inside the JRE libraries / [...]

it worked! thank you, sam.

Sammy1Am, 2024-04-12:
Glad it worked!

I'm actually going to re-open this just in case this *is* some sort of Moppy/Java22 compatibility thing so it can be looked into.  If it's a Java22 bug, it might be fixed (or go away in 23 later this year), but otherwise Moppy might need an update.

Sammy1Am, 2024-07-19:
If you have a moment to try with Java 22 now, @yarfw , There's a [new release](https://github.com/Sammy1Am/Moppy2/releases/tag/2.2.0) that should hopefully address that.

markgrecowork, 2024-07-30:
I also have the same problem with the black screen and no GUI
tried deleting Java and use 22 that failed
deleted everything and started over with no luck.
tried a different computer also no luck

markgrecowork, 2024-07-30:
got it open using Java 18

Sammy1Am, 2024-07-30:
@markgrecowork Were you using 2.2.0 or 2.1.0?

markgrecowork, 2024-07-30:
2.20

Sammy1Am, 2024-07-30:
Hrm.  Well I know you've got it working now, but if you have time to check what the error message was that could be helpful.

Otherwise, I'll give it another try maybe on a new system and figure out what's missing.