samphonic/Moppy2 #211: Using without GUI ?

vallai, 2024-04-26:
Hi !
Is it possible to use the software without GUI ?

I'm planning to use a Raspberry without GUI to control an Arduino, and it would be awesome to be able to start music by launching the bash script with some parameters, or some other way.

Sammy1Am, 2024-04-26:
It's not currently possible out of the box, but *technically* there is no reason it cannot be done.  The [`MoppyLib` folder](https://github.com/Sammy1Am/Moppy2/tree/master/Java/MoppyLib) was intentionally separated from the GUI so that it could be easily embedded in other programs.

You'd have to write a Java program that accepted CLI commands and called the MoppyLib methods to execute though.  I did a quick search on GitHub and came across [this repo](https://github.com/sunsetkookaburra/moppy-utils/tree/main) as well.  It looks like they're trying to send serial commands directly from bash to the microcontroller, which will definitely work, but would require reproducing the functionality of the MoppyLib java code in Bash.

You could also take a look at @DJthefirst 's repo [here](https://github.com/DJthefirst/Mechanical-Midi-Music).  Though I'm not sure if it has a CLI yet either.  It's a new implementation inspired by Moppy, but I believe has a web UI so you could access it from e.g. a cell phone or something if it was hosted on the Raspberry Pi.

DJthefirst, 2024-04-27:
With my software you can use hairless Midi to just convert midi into a serial connection. I assume it's probably callable from CMD. Mine is also currently ESP32 on which I hopefully will get around to changing this before release this summer when finals end in the next few weeks.

vallai, 2024-04-27:
Thanks guys for your answers.

I had already taken a look at [MoppyLib](https://github.com/Sammy1Am/Moppy2/tree/master/Java/MoppyLib), but I think I don't have enough skills to implement a CLI interface easily.

The [Mechanical-Midi-Music](https://github.com/DJthefirst/Mechanical-Midi-Music) repo seems better suited to my project, I think I'll go with it. I hadn't planned on using solenoids but it could be a cool feature.

In the meantime, I also came across the project [https://github.com/wendlers/Moppy](https://github.com/wendlers/Moppy) which could be a good alternative. It's a fork of the [first version of Moppy](https://github.com/Sammy1Am/MoppyClassic) that lets you control floppies directly from a Raspberry Pi.

DJthefirst, 2024-04-27:
Oh did you want to use the GPIO off the pie to play your instruments or just as a server? I can work on porting it which should be pretty quick as only the network, GPIO and and interrupt need to change but it will be a few weeks until after my finals are over.

I can also try making a quick CLI interpreter server-side but you basically just convert midi to serial. You can preprogram the Distributors from the web GUI that are device side. Distributors are equivalent to MoppyDrive mappers which are server side on Sammy's project.

vallai, 2024-04-29:
I don't particularly want to use the Raspberry Pi's GPIOs. My goal is to be able to play music on hardwares, without having to plug it into a computer. Everything should be manageable via a screen and buttons for selecting music, like a jukebox.

A friend gave me an ESP32, and I'm going to try using it on your project, where the server will be running on the Raspberry Pi. If it works, I'll try to start the songs programmatically on the Raspberry Pi via a headless browser like PhantomJS.

DJthefirst, 2024-04-30:
Ok sounds good I'll still work on a CLI interface in the next two weeks that should solve all your problems but one thing to note is the GUI only works on chromium desktop browsers ex: Chrome, Edge, Opera. I haven't tested the pi but IOs Android and Lite versions of these borrowers do not support Web USB which I use to talk to the ESP32. The GUI is only used for config which once set is stored on the esp32 so after it is configured you should be able to use hairless Midi on the PI. You can also choose to manually program the distributors config in main which I can help you with if you're interested. Let me know how I can help as the project is still in dev the before release this summer.

DJthefirst, 2024-05-11:
Quick update to this thread the CLI is mostly done I only spent the last 2 days on it in my free time so it is hideously hacked together as I learn the .NET framework but it is confirmed working on Raspberry PI. Because it is C# .NET it should also work Win/Linux/Mac/Andriod/(Maybe IOS but I doubt they support serial access). The next 2 days are graduation but after that, I should be able to finish it up in the next week. I will be functional but only a 0.5 version as I will rewrite it with proper software engineering as the core for my Desktop GUI. But I figured it would be good to get something functional out for now. Will also be providing room for suggestions and better documentation when I get it more usable next week. 

Also If I didn't already mention it multiple times I am porting my entire project to C# .NET where I can have the same code shared across my WEB, Desktop, and CLI GUIs. 

https://github.com/Sammy1Am/Moppy2/assets/37967524/8ed4983e-23a5-466e-b87a-5bae9db40d29

solidsnake745, 2024-06-15:
Hi, I'm quite late to this thread, but wanted to offer a possible alternative with some caveats.

It's also possible to play MIDI files from an SD card and have buttons and an LCD screen as an interface.
There is an Arduino library for reading MIDI files from an SD card called [MD_MIDIFile](https://github.com/MajicDesigns/MD_MIDIFile). I've combined that with my own library I'm working on to demonstrate: https://photos.app.goo.gl/4KS4miPLkXEMDtDh6

[The code for this sketch is available here for now.](https://github.com/solidsnake745/MIDI_Device_Controller/blob/Development_IO_Refactor/examples/Interface%20Examples/3Buzzers_MD_MIDIFile/3Buzzers_MD_MIDIFile.ino) FYI it doesn't compile for an Uno/Nano as between the two libraries, too much memory is consumed.

I didn't have buttons and a screen to easily add, but I've done this in the past a while back:
https://www.instagram.com/reel/BJdJYDjjqPx/
https://www.instagram.com/reel/BKdrWiCjiGH/

For the caveats:
1. My library is definitely a work in progress and I haven't been working on it for quite some time. There still a number of things I'd like to change.
2. I don't have any real documentation on using it as I was still making changes. My hope was to instead provide example sketches for now that work with the current version that people could copy as a basis. Once things are more complete and not prone to changing I'll be updating the readme/wiki.
3. I still have yet to implement ESP32 support, but it's on the horizon once I finish some other things. Should be a fairly straightforward change.
4. The biggest issue is that my library doesn't run as well on the Arduino Uno/Nano and more so aimed at the newer [Teensy series boards](https://www.pjrc.com/teensy/). The complexity I've introduced definitely slows things down a bit, but I find it hard to justify when more advanced boards are available at the same or even lower cost sometimes.

I have to run my library at a resolution of 100 (as opposed to 40 for Moppy) on an Uno/Nano, but can run it at 20 on a Teensy 3.2 or even 2 on a Teensy 4.0. Unos/Nanos are generally like $25 and the Teensy 4.0 is also like $23m but runs at 600MHz versus 16MHz. I was thinking of possibly creating another lite version later on that is performant but with less features down the line. I figure if that's what someone needs right now they already have alternatives like Moppy available.