samphonic/Moppy2 #212: Error when uploading to Arduino R3

averagenokiauser, 2024-05-09:
Hi, I am getting these errors when trying to upload the code to Arduino Uno R3:
/var/folders/2f/n704ym951790x1np_pr8jrvr0000gn/T//cczhQTfa.ltrans0.ltrans.o: In function `main':
/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/main.cpp:43: undefined reference to `setup'
/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/main.cpp:46: undefined reference to `loop'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
Does anyone know how to fix this?

Sammy1Am, 2024-05-09:
What tools / IDE are you trying to use?  `loop` and `setup` are part of the Arduino framework, so it seems like that might be missing or misconfigured.

Google might be a good source of answers for this one.

averagenokiauser, 2024-05-09:
Thanks for the quick reply, I am using Arduino IDE 2.3.2 on Mac OS. I also tried some older versions of IDE and its the same. I even tried running it on Linux and its also the same. About the tools I installed TimerOne if that's what you mean with tools. Sorry I got the Arduino today so I don't know much about it right now.

Sammy1Am, 2024-05-09:
Did you open *only* the `.ino` file in the IDE, or have you opened other files?  Sometimes it gets weird if you try to open anything but that one file.

averagenokiauser, 2024-05-09:
I opened the Moppy2-Arduino.ino file that is in folder Moppy2-Arduino.

averagenokiauser, 2024-05-09:
Any ideas?

averagenokiauser, 2024-05-09:
Okay, so I tried adding void loop(){} and void setup(){} to the code as someone said and now it seems to upload it without any problems. Will it work okay with this code added or not?

averagenokiauser, 2024-05-09:
When I click the reset button nothing happens. Its probably because of that piece of code I added. Also where exactly do I connect ground wire on the floppy drive? Right now its on pin 31 as on wiki it says to connect it to a odd number pin.

Sammy1Am, 2024-05-09:
> Okay, so I tried adding void loop(){} and void setup(){}

It's weird that this helped as loop() and setup() are already [defined here](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/main.cpp#L65-L80).  When you say "to the code" what file did you add it to?

Did you unzip the release after downloading it and open the `.ino` from the unzipped directory?

It sounds like the code you've added essentially says "do nothing" for setup and loop, and it sounds like that's what it's doing.  Best to remove the changes and figure out why the IDE isn't finding the actual code.

> Also where exactly do I connect ground wire on the floppy drive?
Pin 31 should probably be fine (but you could also use the odd numbered pin across from one of the in-use pins just to be sure).  17 or 19 might be a safer bet in case the particular drive you have isn't using 31 for some weird reason.

averagenokiauser, 2024-05-10:
I added that code to Moppy2-Arduino.ino file. Yes I unzipped the release.

averagenokiauser, 2024-05-10:
[image]
This is how the unzipped folder looks like for me.

averagenokiauser, 2024-05-10:
[image]
Is it normal that I get this error when opening the file?

averagenokiauser, 2024-05-10:
Found the issue! You just need to rename release zip to Moppy2-Arduino after downloading and it will work.

Sammy1Am, 2024-05-10:
Ohhhh, well there you go.  And it was actually like that automatically until 2.1.1 where I guess I messed that up.  I'll see if I can fix the release zip.  Good catch!

Sammy1Am, 2024-05-10:
Fixed the release binary.  Thanks again.  If you get the rest working feel free to close this issue (or come back if there's something else you've got a question about)

averagenokiauser, 2024-05-10:
No problem! Few more questions: do I need to change the config if I wire more than one floppy drive? If yes I then need to upload the code again right? Also can I connect more floppy drives to the same ground pin?

Sammy1Am, 2024-05-10:
Not at my computer, but IIRC the config is set for 8 drives out of the box,
so you're good for at least that many.

All the grounds can be connected together (daisy chained or star should
both work fine for this application)

averagenokiauser, 2024-05-10:
Everything works now. I connected 3 floppy drives and its really awesome. Do you maybe have more midis that are optimized for floppy drives?

Sammy1Am, 2024-05-10:
Glad it's working!

Offhand I do not (I didn't post them online anywhere for IP reasons).  Most MIDIs will work okay with round-robin addressing and a little bit of octave adjustment.  It might be work looking around to see if anyone else that's making Moppy YouTube videos, etc. have shared their MIDIs though.  Good luck!

DJthefirst, 2024-05-11:
Look on YouTube for piano tutorials with midis. Also play with what midi channels your using in the mapper and you will eventually find guitar tracks which tend to sound best in my option. I like to move it sounds good as well as crazy train.

averagenokiauser, 2024-05-11:
How do i select channels in mapper?

Sammy1Am, 2024-05-11:
Each mapper box executes TypeScript (JavaScript) to determine if and where MIDI note events are sent.  The first box returns `true` for notes that that mapper-row should handle.  The default "Channels 1-4" returns true if the MIDI channel is 1-4, but false otherwise.  You can change this formula to select more, fewer, or different MIDI channels.

With one Arduino, the second box will always be `1`.

The third box determines which floppy drive will have the note assigned to it.  The default is that channel 1 goes to drive 1, channel 2 to drive 2, etc. *  But if you're selecting e.g. channel 6 from the MIDI, you'll need to change the formula in the Sub Address box so that it's not assigned to drive number 6 (which you may not have).

The fourth box outputs the note to play.  `n` is just the note from the MIDI file, but you can add or subtract 12 to move the notes up or down an octave (in case whatever instrument is on that track is too high or low for floppy drives).

* I think there's actually some sort of bug there as the conditions are checking for channels 1-4, but then the Sub Address script is doing c+1 which would map to drives 2-5.  So offhand I can't remember if c is zero-indexed or not, but one of those two boxes is wrong :P

averagenokiauser, 2024-05-11:
Oh okay. I for some reason have two mappers. Is that okay? Also on the last box what exactly do I type so it does not play stuff that's too low or too high? Another question: What software are you using to edit midis? I have some songs that would play nicely on floppy drives but i would first need to somehow optimise them so floppies don't play stuff they can't.

Sammy1Am, 2024-05-11:
> I for some reason have two mappers. Is that okay?

It's not inherently a problem, you can remove one or add more if needed.  Just try to avoid sending duplicate notes to the same floppy drive.

> what exactly do I type so it does not play stuff that's too low or too high?

If you want it to *skip* notes that are too high or low, you can add something like `&& (n>40 && n<80)` to the first box (but not those numbers-- I don't remember what the highest and lowest note values are).  If you want to squash notes into a certain range, you can use mod in the last box to reduce to a limited octave range, and then add multiples of 12 to bring it up to the right value (something like (but not exactly): `(n%48) + 24`

> What software are you using to edit midis?

Heh, I mean nothing for years, but back in the day I used Sibelius which is definitely overkill.  AnvilStudio is pretty decent for some light editing.

averagenokiauser, 2024-05-11:
Thanks. Do you have any idea how to wire a motherboard buzzer for it to work? It has only two wires one is + and one is -. I have seen people using it on YouTube but I am not sure if they used your software.

Sammy1Am, 2024-05-11:
So *assuming the Arduino can safely drive the buzzer* (i.e. without a transistor or some), you'd just connect - to ground and + to the even-numbered pin associated with the floppy drive you'll be replacing with the buzzer.

Basically the buzzer will pulse at the same frequency as the stepper motor would step, so you're just connecting it to the step pin.

averagenokiauser, 2024-05-14:
Thanks. it worked. Is it possible to wire scanners and HDDs so they work? I have some dead hard drives and an old canon scanner. I have seen some people using them but there isn't any guide on how to wire them. Also sorry for asking so many questions i just really like this project.

Sammy1Am, 2024-05-14:
No worries, I'm glad you're enjoying it!

So, anything with a stepper motor will technically work (hard drives, scanners, printers, CNC machines, drones, etc.), but stepper motors require some sort of circuitry to control them, so you can't just plug them directly into the Arduino.  Floppy drives just happen to work really easily because the stepper motor driver circuitry accepts 5v on/off signals as input and have that input easily accessible as pins.  For pretty much all of the other options, you'll need to get an appropriate stepper motor driver to connect to the motor in question (or somehow otherwise trick the circuit into working the way floppy drives do).  I unfortunately don't actually have any experience with any of that myself, so you'll have to go hunting.  The answer is definitely "yes" though, but there may be some more work.

One potential exception to this is the read head on harddrives.  The head works basically the same way a speaker does where an electrical current through a loop of wire in a magnetic field moves the head back and forth.  I haven't actually tried this, but I **think** that if you directly connected that loop of wire to the Arduino (maybe with a protective resistor in there to start so you don't kill the Arduino) the head would vibrate a bit like a speaker.  You might need to e.g. put a rubber band on it to bring it back (since the output is just 5v / 0v rather than +5v / -5V), but I *think* that would work.

averagenokiauser, 2024-05-15:
[image]
So I connect wires to some of these pins right?

DJthefirst, 2024-05-16:
I put gnd on one pin and rake 5v across the rest until it clicks you might want to have a small 100ohm resistor for protection though I usually yolo. typically one of the outside pins on that connector is gnd it just takes some trial and error. Or use a multimeter for resistance. Should be 20-400 ohms ish but I haven't measured it myself so I can't say for sure.

averagenokiauser, 2024-05-19:
I found the pins that make it click. Do I connect it to the Arduino the same way as floppy drives? If yes can I somehow select the disk to play drums in the GUI?

averagenokiauser, 2024-05-19:
I have wired it just like I would wire a floppy drive which means ground pin is on ground pin on Arduino and the other pin is on even numbered pin. Its making some noise but it seems like it isn't getting enough power to move the head. Is there some way to make it have more power?

DJthefirst, 2024-05-20:
Not really you would want a transistor wired to a external power supply to push the needed current. Or an H bridge motor driver but I haven't gotten around to trying it yet.

averagenokiauser, 2024-05-23:
> Not really you would want a transistor wired to a external power supply to push the needed current. Or an H bridge motor driver but I haven't gotten around to trying it yet.

How exactly do I push it? Do I just connect it to some pin?