samphonic/Moppy2 #214: Checking COM port in Network Bridges list causes error - Failed to open serialPort

solidsnake745, 2024-06-15:
Hello, I'm trying to run the Moppy2 GUI on Windows 10. I have JDK 21 installed.

In the UI, I see multiple COM ports displayed and some random IP I don't recognize. When I try to connect (check) any of the COM ports, I get the below error and the box isn't checked (of course). Connecting to the IP does work as far as being able to check the checkbox.

```
Jun 14, 2024 10:42:15 PM com.moppy.control.gui.netpanel.BridgePanel bridgeCheckBoxActionPerformed
SEVERE: null
java.io.IOException: Failed to open serialPort!
        at com.moppy.core.comms.bridge.BridgeSerial.connect(BridgeSerial.java:40)
        at com.moppy.core.comms.bridge.BridgeSerial.connect(BridgeSerial.java:19)
        at com.moppy.control.NetworkManager.connectBridge(NetworkManager.java:120)
        at com.moppy.control.gui.netpanel.BridgePanel.bridgeCheckBoxActionPerformed(BridgePanel.java:82)
        at com.moppy.control.gui.netpanel.BridgePanel$1.actionPerformed(BridgePanel.java:57)
        at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
        at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2314)
        at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:407)
        at java.desktop/javax.swing.JToggleButton$ToggleButtonModel.setPressed(JToggleButton.java:411)
        at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)
        at java.desktop/java.awt.Component.processMouseEvent(Component.java:6621)
        at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3398)
        at java.desktop/java.awt.Component.processEvent(Component.java:6386)
        at java.desktop/java.awt.Container.processEvent(Container.java:2266)
        at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4996)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4828)
        at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948)
        at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575)
        at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310)
        at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4828)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:775)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:720)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:714)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:400)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:87)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:98)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:747)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:400)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:87)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:744)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
```

Not sure what's causing it and I sort of feel it's something specific to my computer, but I can't really debug/troubleshoot further.
I tried to debug the source myself, but am also running into compilation issues when using the Netbeans IDE. This is a separate issue, just mentioning what I've tried to get more info in case you might know something. Not really getting a lot of out Google searches on this.

<ins>Info on/from NetBeans</ins>
Product Version: Apache NetBeans IDE 22
Java: 21.0.2; Java HotSpot(TM) 64-Bit Server VM 21.0.2+13-LTS-58
Runtime: Java(TM) SE Runtime Environment 21.0.2+13-LTS-58
System: Windows 10 version 10.0 running on amd64; UTF-8; en_US (nb)

<ins>Output from build</ins>
```
JAVA_HOME="C:\Program Files\Java\jdk-21"
cd C:\Users\solid\Desktop\MoppyControlGUI-2.1.0\Java\MoppyControlGUI; C:\Users\solid\.gradle\wrapper\dists\gradle-8.7-bin\bhs2wmbdwecv87pi65oeuq5iu\gradle-8.7\bin\gradle.bat --configure-on-demand -x check build
Configuration on demand is an incubating feature.
> Task :MoppyLib:compileJava FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':MoppyLib:compileJava'.
> java.lang.IllegalAccessError: class lombok.javac.apt.LombokProcessor (in unnamed module @0xe9178b0) cannot access class com.sun.tools.javac.processing.JavacProcessingEnvironment (in module jdk.compiler) because module jdk.compiler does not export com.sun.tools.javac.processing to unnamed module @0xe9178b0

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

For more on this, please refer to https://docs.gradle.org/8.7/userguide/command_line_interface.html#sec:command_line_warnings in the Gradle documentation.

BUILD FAILED in 199ms
1 actionable task: 1 executed
```

Not familiar with Netbeans + Gradle so I might be missing something obvious.
Thanks for any help.

Sammy1Am, 2024-06-15:
Haha, man, this code is slowly falling apart it seems.  Probably time for a bit of a refresh on the dependencies (especially since it looks like it fails to run in Java 22 now as well).

The error in NetBeans is something about Lombok, which is a... Plugin (I think?) for NetBeans, but the version referenced by the project might be pretty outdated.

I don't know for sure when I'll get time, but on my list is updating all the libraries to recent versions (which I think is probably the root cause of the sporadic serial issues, and the lack of Java 22 compatibility).

Is NetBeans still a good Java IDE?  I might try using VS Code this time around.

solidsnake745, 2024-06-15:
Thanks for getting back to me so fast. Sounds good to me.

I did see mentions of that error when using older versions of the lombok plugin. So I tried to change the version by updating Build Scripts/build.gradle. Specifically changing org.projectlombok:lombok under dependencies to 1.18.22. But that didn't seem to fix anything. I did confirm I saw the version of lombok under Configurations/annotationProcessor and Configurations/compileClasspath change to that version. Other than that I also tried to figure out what version of Gradle Moppy was using versus what Netbeans was using, but I think Netbeans auto-selected the version and downloaded it as far as I can tell.

So I'll just wait for the update. From the little research I did, I hear IntelliJ is a better option than VS code for Java. Apparently you need a lot of extensions to get it working in VS code and even then experience is still not that great. I know nothing firsthand though as it's not something I work in.

Sammy1Am, 2024-07-19:
[New release](https://github.com/Sammy1Am/Moppy2/releases/tag/2.2.0) is out that updates dependencies and resolves at least the issue with Java 22.  If you have a moment to test I'd appreciate the feedback (I'm sort of updating blind as I don't have my floppy drive hardware setup).

solidsnake745, 2024-08-05:
Hey, sorry for getting back to you so late.

I can confirm the new release is working with java 22. It also looks like the issue I was running into is specific to my machine and I didn't realize it earlier.

I have several additional com ports added by my Bluetooth adapter that's part of my motherboard. So the error was caused when trying to connect to one of those. Once I found an actual serial port I was able to connect and play a song. I'm going to close this issue in that case. 

Unfortunately I'm not currently using the moppy Arduino code so I can't confirm that it's actually playing notes from the microcontroller. But I would think that's working as it did before. 

Thanks.

closed 2024-08-05 by solidsnake745

Sammy1Am, 2024-08-08:
Thanks for the update-- yeah I didn't change any of the microcontroller code or the communication with it, so it *should* still be working.