samphonic/Moppy2 #215: How to connect fdd to esp8266

Lord-q, 2024-06-19:
I also have a problem: the wireless connection does not show the connected device but a white And I don't know how to connect fdd to esp8266

Lord-q, 2024-06-19:
[image]
I think I did something wrong?

Sammy1Am, 2024-07-19:
Hey there, sorry this slipped through the cracks.

First, make sure you switched from COM to UDP in [`MoppyConfig.h`](https://github.com/Sammy1Am/Moppy2/blob/d5fcab0a1faf911800786e7691eb7963da4b3879/Microcontroller/Moppy2-Arduino/src/MoppyConfig.h#L27) so that the ESP will connect to UDP and not be waiting on COM input.

I've personally only ever used a shift register with an esp8266, but the pin numbers should work the same, so the first drive would be connected to pins GPIO2 and GPIO3.