samphonic/Moppy2 #216: MoppyControlGUI not starting

CsabaiGD, 2024-07-17:
Hi,

I'm currently having some issues with starting the GUI. When I try to run the .bat file from CMD, it displays the message below. I've already updated the Java, and rebooted my computer, but still the same.
[image]
[image]
Thanks, any help would be appreciated!

Sammy1Am, 2024-07-17:
It appears you're still using Java 8.  Moppy needs... I think like 12 or 13 to run.  But the very latest version breaks things, so stay at like 21 or below for now.

Hopefully I'll get time soon to refresh the code so 23 and up will work.

Sammy1Am, 2024-07-17:
Sorry, should have added: if you installed a version other than 8, you need to make sure your path / JAVA_HOME are updated appropriately.

CsabaiGD, 2024-07-17:
So I tried to install the latest version of Java from the Java control panel (clicked "Update Now"). I can't really understand why and how did it stay at version 8... but now, I searched up specifically for Java 17, installed it and now it works perfectly! 
Thank you very much! :)

Sammy1Am, 2024-07-17:
Glad you got it working!