samphonic/Moppy2 #220: I didn't understand coding well

Reict232, 2024-10-14:
I don’t understand which sketch needs to be loaded into the Arduino for everything to work. It’s interesting to collect, but I don’t have any knowledge. The instructions are not complete.

Sammy1Am, 2024-10-14:
`Moppy2-Arduino.ino` is the sketch that should be opened in the Arduino IDE.  [The wiki](https://github.com/Sammy1Am/Moppy2/wiki/Getting-Started#arduino-ide) instructions should be fairly complete.  Is there a particular step you're getting stuck on?

Reict232, 2024-10-17:
The INO file contains plain text instead of code. I don't speak English well and the translation of the instructions is bad, if you can direct the firmware file I will be very happy :)

Sammy1Am, 2024-10-17:
> The INO file contains plain text instead of code.

This is correct.  The Arduino IDE will compile the code in the `src` directory when you open that sketch.  Please make sure you have unzipped the folder first, before opening it in the Arduino IDE.

I'm sorry the translation is not clear.  Would you mind explaining the steps you have completed so far?  What step are you stuck on?  What is the error/problem you are seeing?

Reict232, 2024-10-18:
I unpacked the archive, opened the ino file, and after loading it into the Arduino, nothing worked for me. I thought it was because the file was empty and wrote a question...
I'll try to download the firmware again

Reict232, 2024-10-18:
Yes, when I downloaded the driver, the firmware loaded. Thanks for the support and for the cool project.