samphonic/Moppy2 #223: Apple 20 pin floppy connection

FloofyProtoBomb, 2024-11-26:
Hello, I am curious if it's possible to connect the older apple floppy drives that just have the single 20 pin cable. I have a box of like 40 of them and want to use them for this.

If anyone has any advice or a pin out for these drives it would be greatly appreciated.

Sammy1Am, 2024-11-26:
Pinout varies a bit based on *which* ~20pin connector; there's a lot of good info on [this page](https://www.bigmessowires.com/2015/03/11/apple-ii-disk-emulation/).

However from a cursory glance, it doesn't look like the apple drives have a `step` or `direction` pin.  Control is probably done via some other sort of (serial?) signal.  That means those drives probably aren't plug and play with the current Moppy.

However it's probably possible to create a new Moppy instrument that emulates whatever communication needed to control the drive-- it just hasn't been done yet.

FloofyProtoBomb, 2024-11-26:
Ah, unfortunately I have no idea how to do that lol. Thanks for the info though!

FloofyProtoBomb, 2024-11-26:
I bet the drives sound really cool though, hope someone does this in the future.