samphonic/Moppy2 #225: Arduino isn't sending to floppies.

Vatiquada, 2024-12-29:
[image]
[image]
[image]
[image]
[image]
[image]
https://github.com/user-attachments/assets/fe7e91a2-da60-4cbe-bfd0-705ead263be8
[image]
[image]

Good afternoon. I built this project in university 12 years ago, and I am looking at getting it going again. I am currently using Arduino IDE 2.3.4
When I open the IDE, I can verify and upload the project with (seemingly) no issues and errors. From there, I use the windows .bat file for MoppyGUI. It did not work in java jdk 23, so I downgraded to 21 and it worked. From there, I select COM1 and change bitrate to 9600 to match the bitrate in device manager. When I add a song and hit play, the arduino lights up the RX light, so I know that something is happening. It just not sending that data to the individual drives.

I have PINS 18/20 connected to ports 2&3 on the arduino. I have PINS 17/19 connected to ground. I know that the documentation does not mention pin 17, though when I did this 12 years ago I have it connected and it worked just fine. 

From the media attached, and from my commentary, can you guys see any issues? I may just rebuild it from the ground up. These floppy drives are tested and do work in PCs, and all 8 lights are turned on. This is the same configuration from 12 years ago, just plugged in and double checked that all wires were still attached.

Vatiquada, 2024-12-29:
As an edit, in the .bat MoppyGUI, when I select COM1 @9600, nothing shows up in the upper right box as mentioned in the instructions, though the RX light does light up. Everything is being run with administrator privileges.

Vatiquada, 2024-12-29:
Update: I read the .h files and chaged the bitrate in both device manager and the GUI to 57600. It now shows up in the GUI software as a device, however, it still does not send to the drives themselves.

Sammy1Am, 2024-12-29:
If the Arduino is showing up in the GUI, the connection is successful and the firmware was uploaded.  This means your issue is either:
### Wiring / the drives.
Make sure you don't have STEP/DIRECTION pins swapped.
Make sure drives are selected (drives with a little LED should have the LED on probably).
Check grounding.
(The second two look fine based on your photo)
### The MIDI file / mapping.
Which MIDI file are you using?
Can you send a screenshot of the GUI with the mappings to make sure they're correct (the defaults should work, but sometimes stuff can get messed up)?

Sammy1Am, 2024-12-29:
Edit: Sorry, I see you already sent a screenshot even (thank you!  Not used to that).

Everything looks okay from the GUI too.  Try swapping STEP/DIR just in case, maybe try connecting directly to one drive just to rule out anything weird with all the drives linked together.  Try a different MIDI file (and also make sure you're waiting a little bit incase the first drive doesn't play notes right at the beginning of the song)

Vatiquada, 2024-12-31:
Hey! Sorry for the late reply. I was able to diagnose it as breadboard sitting for 11 years with a hand soldering job not holding up to the test of time. I took it all apart and re-soldered it. I can now get 2 drives to play the tune when it initializes, though 4 drives are plugged in. I have drive 1 on pins 2&3, drive 2 on pins 4&5, drive 3 on pins 6&7, and drive 4 on pins 8&9.

Only drives one and two play the tone, and when I have midi files sending 4 tracks, only tracks 1&2 play, not tracks 3 and 4. Is there a way to enable more tracks? 5 or 8 would be perfect for me. I remember years ago having 8 drives playing 5 tracks (5 channels). Doubling either the treble or bass for songs, depending.

Sammy1Am, 2024-12-31:
No worries, glad you got it partially working!

By default the code is configured for 8 drives, on pins 2-17.  The startup sound will only play on the first drive (pins 2&3), but you could move some wires around to check that the other drives are working.

MIDI files have a weird relationship between "tracks" and "channels".  In Moppy, each *channel* is (by default) assigned to a separate Sub Address.  Each MIDI *track* may potentially contain note events for multiple channels, and the track index is unrelated to the channel number of its events.  All this to say, just double-check that your MIDI file is using channels for which you have drives assigned.

You could maybe temporarily edit the scripts in the GUI to duplicate the first and second channels to all 8 drives just to make sure they're all working.