samphonic/Moppy2 #227: Using an Arduino mega with 22 drives

creeperita09, 2025-03-14:
While i'm aware that the midi standard only supports 16 tracks i'm planning to make 6 of the drives play what the first 6 drives play so the main melody of the song is louder, i also have figured out how to power them (using an atx power supply), the only thing i'm not sure about is if the setup would work, i've look trough the code and i know what to modify, i'm just wondering if it is possible or if there's some limitation (maybe the cpu of the arduino).

Sammy1Am, 2025-03-14:
I'm not sure I follow your plan.  You want 6 drives for each if the first 6 channels? (I.e. 36 drives). There's only 9 available outputs on the Arduino Uno; you could just wire the drives in 6s if you're okay with them always playing the same notes.  Otherwise you could use a bit shifter to add more outputs, but I'm unsure if you'll be able to write out fast enough to control the drives.

creeperita09, 2025-03-14:
no like i mean i want 16 drives to do the 16 channels midi can handle and the remaining six to play the first 6 channels (i'll do it with a mapper) i was just wondering if the arduino mega is capable of handling 22 drives at once

Sammy1Am, 2025-03-14:
Heh, well that will be limiting...  In theory the Mega has enough GPIO pins to do it.  Whether it has enough processing power depends a bit on the music, but it should *probably* work?

I know supplies could be limited, but just wanted to be clear that using multiple Arduinos is also an option.

creeperita09, 2025-03-14:
Thanks anyways, it's just that i don't really want to buy another arduino (even if i get a cheap one from AliExpress or something like that), I'll reopen this issue if i find that the mega alone isn't enough.

closed 2025-03-14 by creeperita09