samphonic/Moppy2 #228: The GUI won't launch

SomeRandomPerson113, 2025-05-03:
I've been trying to make a floppy project because I found others online do it, but the problem is, the App won't launch, and it gives me this message of "Windows protected your PC" sometimes when I try to run it. Then when I press "run anyway" I get nothing. Is there a way you make make it so It could run like with a file type an app would use? (Eg: EXE, APP, BIN).

Sammy1Am, 2025-05-03:
When you download the zip file, right click, go to "Properties" and check the "Unblock" checkbox and click Okay. *After* you've unblocked it, then try unzipping it and running it.

SomeRandomPerson113, 2025-05-03:
So I did try out unblocking it, and it didn't tell me the "Windows protected your PC" again, but it did not launch the file. I am going to try to see if I can convert the file to .exe or any extension that apps use.

SomeRandomPerson113, 2025-05-03:
Well, I cant convert it. Ima try scripting smth to substitute the app on the Arduino IDE.

Sammy1Am, 2025-05-03:
To clarify, you're trying to double click the `.bat` file, correct?

If you open a command prompt and launch it from there, there might be an error message that will help you.

SomeRandomPerson113, 2025-05-03:
Yes, I was double clicking the .bat file. I'm on my phone right now. Sometime between 4:30 or 5 pm I'll be back to my computer. I was also trying out older versions of the moppy control software. Thank you for your help.

Sammy1Am, 2025-05-04:
Older versions are unlikely to to work, some of the stuff has gone out of date.
If you open the folder containing the `.bat` in Windows Explorer, you can just type `cmd` into the location bar to open a command prompt in that directory.  Then just type the name of the `.bat` and hit enter.  If there's an error it should show in the console.