samphonic/Moppy2 #229: Playback is choppy on an 8-inch FDD

yh2237, 2025-06-09:
What's happening is as stated in the title.
I played the sample song Tetris.midi.
It worked on the 3.5 inch.

Version: Moppy2-2.1.1

https://drive.google.com/file/d/10RyWot4dgMgtPIrbOMJ0d0MiFbBThF7L/view?usp=drive_link

Sammy1Am, 2025-06-10:
Hmmm.  Well first: whoa, cool 8-inch drive 🤩! Thank you for the video too, makes troubleshooting a little easier.

If it worked on the smaller drive, we can rule out software or the firmware installation. It's not cutting out just on high or low notes, it really seems just random, so it's not a frequency issue.

One thing I noticed is that it doesn't seem like the head is travelling all the way forward and backward before reversing direction.  Usually on the 3.5" drives it should move nearly all the way to one end before moving back to the other end, so that seems potentially related.  I don't have any personal experience with 8" drives, but some thoughts:

- Make sure the Arduino and drive's grounds are well connected.  The FDD pinout technically specifies all the odd pins as ground, and in my experience they're all tied together, but maybe confirm that that's true with yours just in case you need to ground all of them separately.
- Likewise make sure all your other connections are secure.  You could test manually by grounding the `STEP` pin repeatedly to see if it steps each time you tap it, or if it skips some.
- 8" drives only have 77 tracks, you'll want to change the `158`s [here](https://github.com/Sammy1Am/Moppy2/blob/d5fcab0a1faf911800786e7691eb7963da4b3879/Microcontroller/Moppy2-Arduino/src/MoppyInstruments/FloppyDrives.cpp#L24) to `152`s.
- The fact that it doesn't appear to be travelling all the way forward and back seems like the best place to start; again, though tedious, maybe try manually moving the head around using the STEP and DIR pins to see if that works reliably.
- It's *pooosible* that there's some sort of current limitation in the Arduino and the drive isn't getting a clean signal, but I'd sort of expect that to happen the whole time instead of just occasionally like it's doing.

Sammy1Am, 2025-06-10:
Amazing!  They're so loud too!

closed 2025-06-10 by Sammy1Am